kenjdavidson/react-native-bluetooth-classic #314: how to make a active conversation betwe

xudo49, 2024-03-07:
**Mobile Device Environment**
On which device(s) would you like this feature to be implemented.   If Android only, the IOS implementations must be created as well - keep not of this for your pull requests.
- Device: [e.g. Android, iPhone6]
- OS: [e.g. API19, Nouget, iOS8.1]

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.  If you've researched the available libraries, feel free to add notes or implementation details as you see fit.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

kenjdavidson, 2024-03-07:
Are there errors?
Anything in the log?
Delimiters setup?

You'll need to do some debugging, but I can't see why this shouldn't work. I just don't have time time or devices to try this out.

xudo49, 2024-03-07:
> Are there errors? Anything in the log? Delimiters setup?
> 
> You'll need to do some debugging, but I can't see why this shouldn't work. I just don't have time time or devices to try this out.

this is not error but I using your example code ( convert to functional ) and try to send message from host ( device 1 ) can be send all of time, opposite in device 2 can't sent a message is disable send button( just hacked by remove disable flag ) is same issue 

[image]

kenjdavidson, 2024-03-07:
I appreciate that you may be using the sample app, but this was never tested with android to android communication.  I did the testing with two very distinct bluetooth devices (they are listed on the readme).  If you're having issues, based on what you've said:

1. Try to debug and see why the `disable` was set to true?
2. If bypassing the disabled flag, and its still not working, try debugging to see why its not working?   I would assume it was disabled for a reason.
3. Is the Second device not sending? Or is the first device not reading?  Both are possible.

Anyhow, as mentioned, you'll need to do some debugging and if there is an issue I can walk you through fixing and accept a PR for it.  But I won't be able to setup and debug this for you.

xudo49, 2024-03-07:
let me try :d

xudo49, 2024-03-12:
@kenjdavidson today I'm just try to make chat using example app I need to know how is work
phoneOne -> call Accept()   <- phoneTwo send some message to phoneOne via writeToDevice() [ send success ]
phoneOne callBack by sendBack via writeToDevice() <- not working. 

question 
if I want to send message back to that devices,  **how to send** or **what method do**
please provide step to procedure like [ disconnect() -> connect() -> writeToDevice() ] for sender
and receiver open accept () thanks

kenjdavidson, 2024-03-12:
The library just wraps the Bluetooth functionality into basic socket programming.  There are two phases that are used:

1. The connection phase.  This is generally used on the device with `connectToDevice()` in order to grab the Socket to a peripheral.  When working with phone to phone, the `accept()` function was added.  So in the grand scheme of things:

```
Phone 1 - `accept()`
Phone 2 - `connectToDevice()`
Phone 1 - `open socket`
Phone 2 - `open socket`
```

2. Once the sockets are opened, it works just like you'd expect:

```
Phone 1 - `write()` will write data to the socket
Phone 2 - `read()` will read data from the socket
Phone 2 - `write()
Phone 1 - `read()`
```

If you want to throw in the listener instead of manual reads:

```
Phone 1 - `onDeviceRead()`
Phone 2 - `onDeviceRead()`
Phone 1 - `write()` will call the Phone 2 `onDeviceRead()` callback
Phone 2 - `write()` will call the Phone 1 `onDeviceRead()` callback
```

There is no need to disconnect and reconnect in order to write.  Once you've `accept` and `connectToDevice` the sockets are open, and should communicate without any issues.

xudo49, 2024-03-12:
> ```
> Phone 1 - `onDeviceRead()`
> Phone 2 - `onDeviceRead()`
> Phone 1 - `write()` will call the Phone 2 `onDeviceRead()` callback
> Phone 2 - `write()` will call the Phone 1 `onDeviceRead()` callback
> ```

I think just try but listener  not working in hooks 

> 
[image]

kenjdavidson, 2024-03-12:
Well that doesn't look right, you're only calling `listening()` in the return from `useEffect()`, which if you read the documentation is the `unloadComponent` function, so it will only start listening when the unload happens?

https://legacy.reactjs.org/docs/hooks-effect.html#example-using-hooks-1

xudo49, 2024-03-12:
> Well that doesn't look right, you're only calling `listening()` in the return from `useEffect()`, which if you read the documentation is the `unloadComponent` function, so it will only start listening when the unload happens?
> 
> https://legacy.reactjs.org/docs/hooks-effect.html#example-using-hooks-1

sorry, wrong pictures :x

kenjdavidson, 2024-03-12:
Sadly I haven't gotten around to re-writing the sample app using functions instead of classes.  But I don't see anything majorly wrong with that code.  

As mentioned before, you'll have to debug the native java code:

- Confirm that the message is being read and stored in the buffer
- Confirm that the delimiter is finding the message and splitting it
- Confirm that it's calling the `sendEvent` method 

if all those things work, then maybe something is up with `useEffect()` which would suck, and I'm sure is unlikely since if so there would have been far more open issues on here.

xudo49, 2024-03-12:
> Sadly I haven't gotten around to re-writing the sample app using functions instead of classes. But I don't see anything majorly wrong with that code.
>
> As mentioned before, you'll have to debug the [...]

I tried Class I worked but, it can't Interact between two phone can you help me double check that code because I'm poor in class base

kenjdavidson, 2024-03-12:
The code looks fine, there's not much I can do without debugging it, so I suggest you spend some time actually debugging it.  If you want to link your repository, I'll take a peak when I can, but I can't promise anything.

xudo49, 2024-03-12:
> The code looks fine, there's not much I can do without debugging it, so I suggest you spend some time actually debugging it. If you want to link your repository, I'll take a peak when I can, but I can't promise anything.

after pull current code. it can't run, [ my env is different ] I need to upgrade react core and something for run these project and Im not sure these lib work fine for a new react and react native version

closed 2024-04-30 by kenjdavidson