kenjdavidson/react-native-bluetooth-classic #317: Important information is missing from the documentation

filippozanfini, 2024-04-24:
It would be very useful to know all the values accepted by each of these parameters (connectionType, charset,...) that can be set during the connection:
https://kenjdavidson.com/react-native-bluetooth-classic/android/device-connection/

Thanks

kenjdavidson, 2024-04-24:
If you feel like the documentation needs a little love, I always more than welcome PRs.   With regards to your specific question:

- `CONNECTION_TYPE` by default allows for `delimited` and `binary` which is fully documented on this page.
- `DELIMITER` is a string, again whatever you want, and documented.
- `DEVICE_CHARSET` this one is a little tricky, as it's mainly related to the `delimited` type and depends on whether IOS or Android, but generally will accept any of the enum types on both.

Once you get into the different connection types `delimited` or `binary` then there are more options that are available, which I believe are documented on:

https://kenjdavidson.com/react-native-bluetooth-classic/android/connection-acceptor/ - which controls accepting connections

https://kenjdavidson.com/react-native-bluetooth-classic/android/connection-connector/ - which controls connecting to waiting devices

Do you have any parameters in specific that aren't listed that you're curious about?

filippozanfini, 2024-04-24:
I have to connect to a bluetooth device and communicate with hexadecimal values (both writing and reading), but I don't know the values that are accepted by the various parameters. For example, reading in the various issues, I noticed that there are doubts about "utf-8" or "utf8" as a value for the charset.

By setting the connection in this way (using Expo) I get this from the bluetooth device: ����

`const device = await RNBluetoothClassic.connectToDevice(address, {
        delimiter: "",
        charset: "utf8",
      });`

closed 2024-04-30 by kenjdavidson