kenjdavidson/react-native-bluetooth-classic #318: Write array buffer to device stopped working

PietroGranati, 2024-04-30:
**Mobile Device Environment**
- Device: ipad Mini
- OS: iPadOS 17.3

**Application Environment**
Provide information about your development environment:
- React Native version: 0.73.6
- RN Bluetooth Classic version: v1.73.0-rc.8

**Describe the bug**
I've upgraded the react native version, and so all the modules.
I were on version v1.60.0-rc.21 before and RN v.0.66
I use this module to send an arraybuffer file to a printer, since the update the file seems to be sent but something is missing and the printer do not start the procedure.
If I create a Buffer variabile with ESC/P code inside it works.

Is it a know issue or something?
I've tryed downgrade to react-native-bluetooth-classic 1.60.0-rc6 but the problem remains

kenjdavidson, 2024-04-30:
I wouldn't be surprised if IOS made more changes that broke things.   They changed the encoding type from `ascii` to `nonLossyAscii` (you can find the issue in the closed issue list regarding that change).  I guess it's possible they changed something again in IOS 17.3.

I would need substantially more information and code to give you any more pointers, but if you have it working by adding your custom variable in ESC/P, then why is this not a way to move forward?  

You'll have to do a little debugging on your end.

PietroGranati, 2024-05-02:
Thank you for the support!
Sorry for the late answer, the custom variable is an option but I'll lost the ability to print images that I generate trought a library tha I have on a server.
Actually I receive the bynary data to send from my server and write it directly to the printer

PietroGranati, 2024-05-02:
I'm watching the issue you've told me about but not sure what I need to check, maybe a I need to try a different encoding from my php page

PietroGranati, 2024-05-02:
Ok I've to admit that I were using connect without parameters I checked your well structured guide and added this:
`DEVICE_CHARSET: Platform.OS === "ios" ? 1536 : "utf-8",`
Now it works again, 1536 is utf-8 ? I cannot find the "specific implementation for details"

kenjdavidson, 2024-05-02:
> Ok I've to admit that I were using connect without parameters I checked your well structured guide and added this: `DEVICE_CHARSET: Platform.OS === "ios" ? 1536 : "utf-8",` Now it works again, 1536 is utf-8 ? I cannot find the "specific implementation for details"

Glad things worked.  I can't find the original Swift documentation for CharSets and `String.Encoding` where that's documented.  Swift is annoying.

PietroGranati, 2024-05-02:
No worries! thank you for the help!!