kenjdavidson/react-native-bluetooth-classic #320: Getting Error while connect device CODE: "EUNSPECIFIED"

krunalshoppiko, 2024-05-03:
I'm Trying to connect Bluetooth Barcode Scanner device in Android but not getting success.

I'm using OnePlus7T for testing

package.json
```
"react": "18.2.0",
"react-native": "0.74.1",
"react-native-bluetooth-classic": "^1.73.0-rc.8"
```

AndroidManifest.xml

```
   <manifest xmlns:android="http://schemas.android.com/apk/res/android">

    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN"/>
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT"/>
    <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/> 
```

build.gradle
```
buildscript {
    ext {
        buildToolsVersion = "34.0.0"
        minSdkVersion = 23
        compileSdkVersion = 34
        targetSdkVersion = 34
        ndkVersion = "26.1.10909125"
        kotlinVersion = "1.9.22"
    }
    repositories {
        google()
        mavenCentral()
    }
    dependencies {
        classpath("com.android.tools.build:gradle")
        classpath("com.facebook.react:react-native-gradle-plugin")
        classpath("org.jetbrains.kotlin:kotlin-gradle-plugin")
    }
}

apply plugin: "com.facebook.react.rootproject"

```

1 )  call  for getting unpaired device list and get the unpaired devices.

    `RNBluetoothClassic.startDiscovery();  // successfully getting device list`
    
2 )  call   `RNBluetoothClassic.accept({});`

3 )  form unpaired device list i select Bluetooth Barcode Scanner device and try to pair

```
     try {
          var currentPairDevice = await RNBluetoothClassic.pairDevice( unpaired.id);
          console.log('pairDevice to device:', 'Bluetooth Barcode Scanner');
          getConnectedDevice(currentPairDevice);
        } catch (error) {
          console.error('pairDevice error:', error);
        }
      const getConnectedDevice = async (pairedDevice: any) => {
          console.log('getConnectedDevice>>>>', pairedDevice);
          try {
            const device = await RNBluetoothClassic.getConnectedDevice(
              pairedDevice.id,
            );
            console.log('getConnectedDevice', device);
          } catch (err) {
            console.log('getConnectedDevice error', err);
          }
        };
```

 while calling above function i receive the pair popup for pair the device . after pair the devices my device is connected with bluetooth but when  i'm calling the  getConnectedDevice it's give me an error **code: "EUNSPECIFIED"**  

[image]

When i try to connect device using below code  it's also give me  an error  

```
const connectDevice = async (device: any) => {
    console.log('connectDevice>>>>', device);
    try {
      const messages = await device.connect({
        CONNECTOR_TYPE: 'rfcomm',
        DELIMITER: '\n',
        DEVICE_CHARSET: Platform.OS === 'ios' ? 1536 : 'utf-8',
      });
      console.log('>>>connectDevice>>', messages);
    } catch (error) {
      console.log('>>error>>connectDevice>>>', error);
    }
  };

```
[image]

how can i solve this error?

kenjdavidson, 2024-05-04:
1.  Get connected device is failing since you're not connected. Paired is paired, connected is an open socket. So the first error is expected.

2. The second error looks like your device is not accepting connections. Either it's the wrong device or it's not in accept more.

You'll need to debug more to find out which.

What is connection type rfcomm? The standard are delimited and binary? Did you write your own connection?

krunalshoppiko, 2024-05-04:
> 1. Get connected device is failing since you're not connected. Paired is paired, connected is an open socket. So the first error is expected.
> 2. The second error looks like your device is not [...]

for connection type i use this type as pr the document [Link](https://kenjdavidson.com/react-native-bluetooth-classic/react-native/rn-bluetooth-device/)

So my device is socket mobile S700 [Link](https://socketimagescdn.azureedge.net/cdn/docs/default-source/downloads/programming-guides/d700-s700-d800-s800-ds800-programming-guide.pdf?sfvrsn=b7c14c5d_25)

When i paired this device it's connect to my mobile bluetooth when i open the mobile bluetooth then i see that device is connected  but i return to my app and try to get connected device or get status of connected device like isConnected using address it's getting error even though device connected with mobile bluetooth.

if my device is connected to my mobile then why i'm not getting is connected status true?
what i'm doing wrong? 

can you help me to connect this device to my app or provide me the steps or code to connect this device.

kenjdavidson, 2024-05-04:
Like I said before.

Paired is paired.
Connected is an open socket.

Your devices is paired to the phone but not connected.

I don't know, you need to debug. I don't have your device or code.

closed 2024-06-10 by kenjdavidson