kenjdavidson/react-native-bluetooth-classic #324: IOS: Request to have the check in write function if the data is Buffer data

duc-tran-epam, 2024-06-11:
**Mobile Device Environment**
- Device: any
- OS: IOS

**Is your feature request related to a problem? Please describe.**
Currently, the app sends Buffer data. However, the write function tries to convert it to a string, causing a failure check. As a result, the data isn't sent to the Printer device.

**Describe the solution you'd like**
I suggest removing the check `if let sending = String(data: data, encoding: self.encoding)`
or implementing another check if the data is Buffer data.

I tried to remove the condition and it's working correctly.

```swift
func write(_ data: Data) -> Bool {
    if let sending = String(data: data, encoding: self.encoding) {
        NSLog("(BluetoothDevice:writeToDevice) Writing %@ to device %@", sending, accessory.serialNumber)
        NSLog("(BluetoothDevice:writeToDevice) 2222")
        outBuffer.append(data)
        writeDataToStream((session?.outputStream)!)
    } else {
        NSLog("(BluetoothDevice:writeToDevice) Writing false")
        return false
    }

    return true
}
```

**Additional context**
Here is the buffer data we passed to write function 

```javascript
      const n = 42; // Feed amount
      const PRINT_AND_FEED_PAPER_COMMAND = '\x1B\x6A';

      const command = Buffer.from(PRINT_AND_FEED_PAPER_COMMAND, 'binary');
      const values = Buffer.from([n]);
      const combinedCommand = Buffer.concat([command, values]);
      await classicDevice.write(combinedCommand);
```

kenjdavidson, 2024-06-11:
Feel free to open a PR, not sure what check you removed, since the "working" code you posted has the line/content that you mention you want to remove above.   I'm not entirely sure why it was done that way, but it does look a little weird that I'm converting to the `sending` string but then never using that string.

trilloc, 2024-12-21:
Hi, is this resolved please? I am trying to send ATZ\r to an ELM327 device.. and that doesnt work

Thank you very much for your fantastic package!
Rajesh

kenjdavidson, 2024-12-22:
Not sure it's resolved, the last comment was that the guy should be opening a PR if he believes he's resolved the issue.

duc-tran-epam, 2024-12-23:
Hi.

it's resoled for me on IOS. I tried to send buffer data instead of string.

Regards,
Duc

DUC TRAN

Lead Software Engineer

Office: +60 3 9212 1998<tel:+60%203%209212%201998;ext=69823>x<tel:+60%203%209212%201998;ext=69823>69823<tel:+60%203%209212%201998;ext=69823>  Cell: +84 9 3859 9134<tel:+83%209%203859%209134>  Email: ***@***.******@***.***>

Ho Chi Minh City,  Vietnam  epam.com<http://www.epam.com/>

CONFIDENTIALITY CAUTION AND DISCLAIMER
This message is intended only for the use of the individual(s) or entity(ies) to which it is addressed and contains information that is legally privileged and confidential. If you are not the intended recipient, or the person responsible for delivering the message to the intended recipient, you are hereby notified that any dissemination, distribution or copying of this communication is strictly prohibited. All unintended recipients are obliged to delete this message and destroy any printed copies.