kenjdavidson/react-native-bluetooth-classic #326: [Android] How to read data from bluetooth by using this repo?

CannoChen, 2024-06-30:
My code is provided below:
`
    const [isConnected, setIsConnected] = useState(false);
    const [peripheral, setPeripheral] = useState(null);  
    useEffect(() => {
        RNBluetoothClassic.isBluetoothEnabled().then(() => {
            console.log('open！');
        });
        RNBluetoothClassic.onBluetoothEnabled(() => {
            console.log("open success！");
        });
        RNBluetoothClassic.onBluetoothDisabled(() => {
            console.log("closed！")
        });
        const readDataInterval = setInterval(readData, 100);
        return () => clearInterval(readDataInterval);
    }, []);

    const connectToPeripheral = async () => {
        try {
            const granted = await requestBluetoothPermission();

            if (!granted) {
                throw new Error(`Access fine location was not granted`);
            }

            // this.setState({ discovering: true });

            try {
                const unpaired = await RNBluetoothClassic.startDiscovery();
                for(const device of unpaired) {
                    console.log(device.name);
                    if (device.name === 'HC-05') {
                        const connectResult = await device.connect();
                        if (!connectResult)
                            throw new Error('connect failure！');
                        else {
                            setIsConnected(true);
                            setPeripheral(device);
                            RNBluetoothClassic.cancelDiscovery().then(() => {
                                console.log("connect success！")
                            });
                        }
                        break;
                    }
                }
            } finally {
                // this.setState({ devices, discovering: false });
            }
        } catch (err) {
            console.log(err);
        }
    }
    const readData = () => {
        // console.log("read data");
        if (!isConnected)
            return false;
        // if (peripheral?.available()) {
        //     console.log(peripheral?.read());
        // }
        return true;
    }`

My question is, once I have successfully connected to Bluetooth, how should I consistently read the data that Bluetooth transmits to me? Thanks for your kindness help!!!!

kenjdavidson, 2024-06-30:
There's 2 methods. Setting up the subscription or repeatedly calling read.

The sample app library has both, the project is a sibling to this one under the main github. (react-native-bluetooth-classic-apps) You should be able to follow that. Let me know if any issues.

kenjdavidson, 2024-06-30:
Oh it look slike you have a timeout with read data. That function takes a data object as a parameter. You're missing that.

closed 2024-06-30 by kenjdavidson