kenjdavidson/react-native-bluetooth-classic #328: Incorrect package="kjd.reactnative.bluetooth"

SentoMachina, 2024-07-13:
**Mobile Device Environment**
- Device: Android
- OS: Android 14

**Application Environment**
- React Native version: 0.73.6
- RN Bluetooth Classic version: 1.73.0-rc.10

**Describe the bug**
A few days ago the build process was going normally without any problems. But now, when building the apk using eas, I encounter the problem as below. I try to change version of gradle but not working. Can anyone help me to fix this, Thinks!!!
[RUN_GRADLEW] Execution failed for task ':react-native-bluetooth-classic:processReleaseManifest'.
[RUN_GRADLEW] > A failure occurred while executing com.android.build.gradle.tasks.ProcessLibraryManifest$ProcessLibWorkAction
[RUN_GRADLEW] >
[RUN_GRADLEW] Incorrect package="kjd.reactnative.bluetooth" found in source AndroidManifest.xml: /private/var/folders/35/xgq4zdtn77nch93146q1_bqm0000gn/T/eas-build-local-nodejs/64789d92-ac00-46ab-a29f-57fb819c69ed /build/node_modules/react-native-bluetooth-classic/android/src/main/AndroidManifest.xml.
[RUN_GRADLEW] Setting the namespace via the package attribute in the source AndroidManifest.xml is no longer supported.
[RUN_GRADLEW] Recommendation: remove package="kjd.reactnative.bluetooth" from the source AndroidManifest.xml: /private/var/folders/35/xgq4zdtn77nch93146q1_bqm0000gn/T/eas-build-local-nodejs/64789d92-ac00-46ab-a29f -57fb819c69ed/build/node_modules/react-native-bluetooth-classic/android/src/main/AndroidManifest.xml.

kenjdavidson, 2024-07-13:
Probably the last pr. Revert the version until it gets reverted.

kenjdavidson, 2024-07-13:
I reverted the change in v10 to get things going. Will reapply it to a different minor version for those that need.

closed 2024-07-13 by kenjdavidson

SentoMachina, 2024-07-13:
Thank u so much. After i updated to new version i can build it.