kenjdavidson/react-native-bluetooth-classic #330: onDataReceived event handler is not cleared when the device is connected again?

ShaneZhengNZ, 2024-07-15:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Android
- OS: Android 10

**Application Environment**
Provide information about your development environment:
- React Native version: 0.73.6
- RN Bluetooth Classic version: 1.73.0-rc.10

**Describe the bug**
So, here is my logic:

I want to connect to Device (say, device A), then subscribe to onDataReceived. If the device disconnects (for example, it is turned off), I will recursively try to connect to the device, until the device is turned on and connected, then I want the onDataReceived to subscribe again.

This logic works, in the sense that I can get the data at the initial connection, then automatically connect to the device when it is turned off and on again.

However, it seems that after the device is turned back on, all the data that I get are duplicated.

If I keep doing that (turn the device off and on again), every time when the device is turned on, I get 1 additional set of the same data.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
When the device is reconnected again, I should only get the data once, not 2, 3, 4, etc times.

I am using a Garmin Glo 2 GPS receiver as an example. The data frequency is 10 Hz (i.e. I should get 10 data strings every second), but as soon as I turn off and on the device, I get 20Hz, 30Hz, etc.

kenjdavidson, 2024-07-15:
The expectation here is that you're maintaining the connection and removing the listener when the connection is dropped.  I don't want to start doing things for people within the library and having it break others expectations, I also don't think the library should be responsible for managing things like this, for example if you're connecting to a number of devices at one time, now the library needs to manage all those?  What about for those that use the `setTimeout` and read manual method?

closed 2024-07-15 by kenjdavidson

ShaneZhengNZ, 2024-07-15:
@kenjdavidson lol. Sorry. I forgot to save the reference to the listener and call .remove when I detect the device is disconnected. Thank you