kenjdavidson/react-native-bluetooth-classic #333: [help] cannot read sent data when you use connectedDevice.read()

brianwachira, 2024-07-24:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Samsung Galaxy A30S, Infinix SMART 7HD
- OS:  ANDROID 11. ANDROID 12

**Application Environment**
Provide information about your development environment:
- React Native version: 0.74.3
- RN Bluetooth Classic version: ^1.73.0-rc.12

**Describe the bug**
Hi there, I have an offline-first mobile app that should be able to share api data from one mobile device to another device that has no access to internet.
I have managed to connect devices through the app with the device.connect() but when I send data ( stringified json ) it is never received. Data is always null
[image]

A sample of the data I am sending to the other device : `{"gps":{"lat":-1.3729792,"lng":36.9360896},"survey_id":"9c74c0be-e671-466b-ab9d-725c9003cbe3","question1":false,"question2":true}`

**Expected behavior**
A clear and concise description of what you expected to happen.
- When I run `connectedDevice.write(data);` it should read the data

**Additional context**
- Maybe the issue could be the delimiter? Maybe I am passing the wrong delimitter?
useBluetooth hook ( some functions )

`
function useBluetooth(){

	const scanForDevices = async () => {
		try {
			const devices = await RNBluetoothClassic.startDiscovery();
			console.log(devices);
			setAllDevices(devices);

			const paired = await RNBluetoothClassic.getBondedDevices();
			console.log(paired);
			setPairedDevices(paired);
		} catch (error) {
			console.error("Failed to scan for devices:", error);
		}
	};
	

	const connectToDevice = async (device: BluetoothDevice) => {
		try {
			const connected = await device.connect({ delimiter: '\n',DEVICE_CHARSET: Platform.OS === "ios" ? 1536 : "utf-8",READ_SIZE: 800000});
			if (connected) {
				setConnectedDevice(device);
			}
		} catch (error) {
			console.error("Failed to connect to device:", error);
		}
	};

	const acceptConnections = async () => {
		try {
			const device = await RNBluetoothClassic.accept({});
			if (device) {
				setConnectedDevice(device);
			}
		} catch (error) {
			console.error("Failed to connect to device:", error);
		}
	};

	const sendData = async (data: string) => {
        //console.log(data)
		if (connectedDevice) {
			try {
				const senddata= await connectedDevice.write(data);
                console.log("send data",senddata)
			} catch (error) {
				console.error("Failed to send data:", error);
			}
		} else {
			console.log("No device connected to send data");
		}
	};
}

export default useBLE;
`

How I detect data change on imports screen

[image]

- Maybe the issue could be the delimiter? Maybe I am passing the wrong delimitter?

manhdev0901, 2024-07-25:
I have same issue.

brianwachira, 2024-07-25:
> I have same issue.

Could you try to share your implementation we see whether we can compare notes?

kenjdavidson, 2024-07-25:
Chances are this is a `delimiter` setting, for example:
- on connection you're using `\n` 
- on accept, you're not setting `delimiter` so it's using the default, which may be `\r\n`?

I forget to be honest.  This should be a fairly straight forward thing to debug in the connection and parsing of data on the native side.  I suggest you debug through Android Studio and check out:

1. whether the data is being received
2. whether the data is being chopped into messages based on delimiter

This will give you the best path to finding out what's wrong.

If you're using the manualmethod of `device.read()` you should probably be using the `device.available()` to see whether there are messages available.

kenjdavidson, 2024-07-25:
Again, you need to debug it yourself to see what's going on, I can't debug your application and the Bluetooth functionality is very specific to your devices.  I'm always happy to help, but just posting code (that usually looks good to the eye) doesn't help me much.

viratishah, 2024-07-25:
> Again, you need to debug it yourself to see what's going on, I can't debug your application and the Bluetooth functionality is very specific to your devices. I'm always happy to help, but just posting code (that usually looks good to the eye) doesn't help me much.

Hi, Thank you, I was able to solve this issue. Adding CONNECTION_TYPE = 'delimited' and using encoding='hex' while calling writeToDevice helped.

I get the value now, but it is some unicode string. 
My data is supposed to be in Hex (using HC-05 fyi). 
I tried changing the receivedData function (as mentioned in doc and #50 ), but it led to the function not returning any value whatsoever.  Any help?

brianwachira, 2024-07-25:
> Hi, Thank you, I was able to solve this issue. Adding CONNECTION_TYPE = 'delimited' and using encoding='hex' while calling writeToDevice helped.
>
> I get the value now, but it is some unicode string. [...]

Wow Could you share small code snippet on how you did yours? where do I add the `CONNECTION_TYPE = 'delimited' and using encoding='hex' `? It may help with mine

kenjdavidson, 2024-07-25:
If you're using hex/binary data, you may want to look at the following:
- https://kenjdavidson.com/react-native-bluetooth-classic/guides/hex-encoding-decoding/
- https://kenjdavidson.com/react-native-bluetooth-classic/android/device-connection/#bytearraydeviceconnectionimpl

They may assist in the data conversion or reading.

brianwachira, 2024-07-25:
> Chances are this is a `delimiter` setting, for example:
>
> * on connection you're using `\n`
> * on accept, you're not setting `delimiter` so it's using the default, which may be `\r\n`?
>
> I forget to be [...]

It finally worked!
I set `delimiter` on accept.

here

[image]

But I must mention @kenjdavidson , the docs doesn't mention about adding connector options on accept
[image]
I think this is something that has to be made clear?

kenjdavidson, 2024-07-25:
Glad it's working.  

With regards to your comment, I hear what you're saying, but if you look at the documentation (at least htis was my logic):

- connector 
- accepter
- connection

Delimiter is a `connection` parameter and is documented on that page https://kenjdavidson.com/react-native-bluetooth-classic/android/device-connection/

I guess it's only cause I'm used to it that I know that the `connection` arguments are passed through the `connector` or `accepter`.  

Feel free to write up whatever you think makes more sense and open a PR.

closed 2024-07-25 by kenjdavidson

viratishah, 2024-07-26:
> With regards to your comment, I hear what you're saying, but if you look at the documentation (at least htis was my logic):
> 
> * connector
> * accepter
> * connection

Okay, finally its resolved. 
Just changed CONNECTION_TYPE to binary and converted the output into buffer object.
`````
receivedData = Buffer.from(event.data, 'base64');
/* this line in case you need a hex string
const hexString = receivedData.toString('hex'); */
`````
Thanks @kenjdavidson for this amazing library and the links as well.