kenjdavidson/react-native-bluetooth-classic #334: New Arch Support

brianwachira, 2024-07-31:
**Mobile Device Environment**
Android & IOS

- React Native relseaed a [new architecture](https://reactnative.dev/docs/the-new-architecture/) 
- React Native team is planning to enable the New Architecture by default by the end of [2024](https://reactnative.dev/docs/the-new-architecture/)

Is there a plan for supporting the new architecture?
You can check out more information on enabling new arch [here](https://github.com/reactwg/react-native-new-architecture)

Would you need help with this? I can spare sometime just to help out in the contribution of this.

kenjdavidson, 2024-07-31:
I don't work for the company I did while creating this library, so I haven't done mobile development in a few years now. If someone would like to contribute I would definitely accept PRs, as I do continue to help out.

But I have zero plans at this point to do it myself. I don't have the time, nor the experience with react native at this point to realistically do it.

brianwachira, 2024-08-09:
> I don't work for the company I did while creating this library, so I haven't done mobile development in a few years now. If someone would like to contribute I would definitely accept PRs, as I do [...]

I see. I'd be interested to contribute but with guidance. If it's ok, could you pin this issue at the issues tab so that anyone else can see it and jump in to comment and contribute?

kenjdavidson, 2024-08-09:
You're more than welcome to contribute, there are a couple things that are important for me to understand though, just based on the requirements.

- I'm assuming that this will actually be a non breaking change, the API will not change, only the Native Module side being converted to a TurboModule.  It's not a big deal, but if it's a breaking change we'll need to cut 0.74x maybe to handle this and determine what is the lowest version of React Native supported.

The new architecture is:
- https://github.com/reactwg/react-native-new-architecture/blob/main/docs/turbo-modules.md

ShaneZhengNZ, 2024-08-17:
My 2 cents - if time is an issue, then, we should do the 1st step to make sure this lib is compatible with the interop layer, as described below:
https://github.com/reactwg/react-native-new-architecture/blob/main/docs/enable-libraries.md

By doing so, the lib users can upgrade to the new RN version when the new architecture becomes the default.

Then we can work on the conversion to the turbo modules.

brianwachira, 2024-10-28:
> My 2 cents - if time is an issue, then, we should do the 1st step to make sure this lib is compatible with the interop layer, as described below: [...]

Ok, New Arch is enabled by default in the current React Native version 0.76
@ShaneZhengNZ could we start? if it is ok, kindly guide

kenjdavidson, 2024-11-21:
Not in my end. I'm looking for help with this. Someone with react native experience and time to do so.   I'm more than willing to assist, review PRs and handle the deployments. But I'm still in the same boat, I done have time to do the development myself.

mojjominion, 2024-11-26:
Hi, I can try and add support for new architecture here at-least for android side of things. I am not aware of ios related stuff.

kenjdavidson, 2024-11-26:
It should be pretty straight forward, from what I remember we don't use any of the things that need to be changed (keys, spread, etc).  SO it should just be testing against 0.74 and adding the new file/options:

https://gist.github.com/cipolleschi/82b7a9561b8861330efabbd3eb08c6f5

mojjominion, 2024-12-17:
I was able to use react native `0.76.3` and `newArchEnabled:true`  with bridgeless and bridge mode both. 
Following 3 apis were tested assuming other things would also work.
- getBondedDevices
- connectToDevice
- onDataReceived

kenjdavidson, 2024-12-17:
The `react-native-bluetooth-classic-apps` repository is where I had different version apps for testing.  Just to keep this clean of example applications.  The issue is that it uses local linking, which caused a problem in older versions of React Native, maybe this is fixed.

Feel free to put your example app into that repo, if you want to share it, so that others can use/test against something common.

From what I remember reading, there shouldn't be too much within the module itself that needs to be migrated.  I appreciate you taking some time to look into using it with the new version of React Native.

mojjominion, 2025-01-07:
Hi @harsh-logixits 

I created a fresh app to test this out, you can check the commit [here](https://github.com/mojjominion/react-native-bluetooth-classic/commit/b753bb416dc9bdf71751b4068f562ffc7b908a4b). See if this helps, there was no change required from the module to work with newArch.

agungtriu, 2026-01-13:
Hi @kenjdavidson ,
any updates on New Arch support?