kenjdavidson/react-native-bluetooth-classic #335: RSSI always empty

avanmalleghem, 2024-07-31:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Android
- OS: Android 13

**Describe the bug**
RSSI is always empty when checking it on a connected device.

**Expected behavior**
I would like to get the RSSI value.

kenjdavidson, 2024-07-31:
From my understanding this is not possible.  The rssi is only populated from the Bluetooth adapter during search/discovery. I don't believe it's available otherwise.

If you feel otherwise, feel free to post a pr. But I think this was asked and confirmed a while ago.

avanmalleghem, 2024-07-31:
Yep, I saw this thread aswell : https://github.com/kenjdavidson/react-native-bluetooth-classic/issues/90
It seems complicated on Android and... Impossible on IOS. Too bad... Thanks for the answer !

closed 2024-07-31 by avanmalleghem