kenjdavidson/react-native-bluetooth-classic #336: RECEIVER_EXPORTED or RECEIVER_NOT_EXPORTED should be specified when a receiver isn't being registered exclusively for system broadcasts

trungphammycolor, 2024-08-05:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Google Pixel (Emulator)
- OS: Android 14 (API level 34)

**Application Environment**
Provide information about your development environment:
- React Native version: 0.73.0
- RN Bluetooth Classic version: 1.73.0-rc.12

**Describe the bug**
The app shut down immediately
**To Reproduce**
Steps to reproduce the behavior:
Build and run app
**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]
Android studio log:
```
FATAL EXCEPTION: main
                                                                                                    Process: com.pangaea.app, PID: 12295
                                                                                                    java.lang.RuntimeException: Unable to get provider com.deploygate.sdk.DeployGateProvider: java.lang.SecurityException: com.pangaea.app: One of RECEIVER_EXPORTED or RECEIVER_NOT_EXPORTED should be specified when a receiver isn't being registered exclusively for system broadcasts
                                                                                                    	at android.app.ActivityThread.installProvider(ActivityThread.java:7770)
                                                                                                    	at android.app.ActivityThread.installContentProviders(ActivityThread.java:7276)
                                                                                                    	at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6983)
                                                                                                    	at android.app.ActivityThread.-$$Nest$mhandleBindApplication(Unknown Source:0)
                                                                                                    	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2236)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:106)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:205)
                                                                                                    	at android.os.Looper.loop(Looper.java:294)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8177)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:971)
                                                                                                    Caused by: java.lang.SecurityException: com.pangaea.app: One of RECEIVER_EXPORTED or RECEIVER_NOT_EXPORTED should be specified when a receiver isn't being registered exclusively for system broadcasts
                                                                                                    	at android.os.Parcel.createExceptionOrNull(Parcel.java:3057)
                                                                                                    	at android.os.Parcel.createException(Parcel.java:3041)
                                                                                                    	at android.os.Parcel.readException(Parcel.java:3024)
                                                                                                    	at android.os.Parcel.readException(Parcel.java:2966)
                                                                                                    	at android.app.IActivityManager$Stub$Proxy.registerReceiverWithFeature(IActivityManager.java:5684)
                                                                                                    	at android.app.ContextImpl.registerReceiverInternal(ContextImpl.java:1852)
                                                                                                    	at android.app.ContextImpl.registerReceiver(ContextImpl.java:1792)
                                                                                                    	at android.app.ContextImpl.registerReceiver(ContextImpl.java:1780)
                                                                                                    	at android.content.ContextWrapper.registerReceiver(ContextWrapper.java:755)
                                                                                                    	at com.deploygate.sdk.DeployGate.prepareBroadcastReceiver(DeployGate.java:273)
                                                                                                    	at com.deploygate.sdk.DeployGate.<init>(DeployGate.java:238)
                                                                                                    	at com.deploygate.sdk.DeployGate.install(DeployGate.java:491)
                                                                                                    	at com.deploygate.sdk.DeployGate.install(DeployGate.java:464)
                                                                                                    	at com.deploygate.sdk.DeployGateProvider.onCreate(DeployGateProvider.java:15)
                                                                                                    	at android.content.ContentProvider.attachInfo(ContentProvider.java:2619)
                                                                                                    	at android.content.ContentProvider.attachInfo(ContentProvider.java:2589)
                                                                                                    	at android.app.ActivityThread.installProvider(ActivityThread.java:7765)
                                                                                                    	at android.app.ActivityThread.installContentProviders(ActivityThread.java:7276) 
                                                                                                    	at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6983) 
                                                                                                    	at android.app.ActivityThread.-$$Nest$mhandleBindApplication(Unknown Source:0) 
                                                                                                    	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2236) 
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:106) 
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:205) 
                                                                                                    	at android.os.Looper.loop(Looper.java:294) 
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8177) 
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method) 
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552) 
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:971) 
                                                                                                    Caused by: android.os.RemoteException: Remote stack trace:
                                                                                                    	at com.android.server.am.ActivityManagerService.registerReceiverWithFeature(ActivityManagerService.jav [... cut ...]
                                                                                                    	at com.android.server.am.ActivityManagerService.onTransact(ActivityManagerService.java:2720)
                                                                                                    	at android.os.Binder.execTransactInternal(Binder.java:1339)
                                                                                                    	at android.os.Binder.execTransact(Binder.java:1275)
```
**Additional context**

kenjdavidson, 2024-08-05:
Did you happen to google that? There are a number of stack overflow articles about it.

You haven't provided
- a description of what you were attempting
- sample code
- have you found any other resources
- etc

Looks like it may be library unrelated. Have you attempted to set that context in your app?

image.png (view on web)<https://github.com/user-attachments/assets/b4a4a4c6-2ddf-4ddb-9270-f4c79ff4ef00>

kenjdavidson, 2024-08-05:
After a few more looks it does look like I can add it to the library. But I'll need you to confirm adding it resolves the issue, then open a pr for it, if you don't mind.

bguerinActidel, 2024-09-17:
Hello,

Any news of a potential fix ? It has been fixed in other library I use but not this one.

Thank you

kenjdavidson, 2024-09-19:
I haven't looked into it, I thought you were going to resolve an open a pr? Since you know how to fix it.

harriha, 2024-12-04:
Hi,

I'm currently in process of updating an app which uses this lib to Android 14, and was expecting to hit this issue - but didn't, things seem to work fine in my case (acceptance testing still ongoing, though). Not sure why, perhaps I'm using the library differently than others above or then there's something different with my test devices. If there's some more details about how to trigger this issue, I'd appreciate the hints.

Anyway, recording an observation here, in case it helps: As I was trying to understand the exact scenario here (I'm not very experienced with Android native stuff), I came to the conclusion that the received intents this lib is registering to are all [system broadcast actions](https://developer.android.com/develop/background-work/background-tasks/broadcasts#system-broadcast-actions), which are apparently [exceptions to the new rules](https://developer.android.com/about/versions/14/behavior-changes-14#system-broadcasts). However, as there's folks in this thread who have experienced an issue with this piece of OS changes, I'm a bit puzzled, might well be that my limited Android skills fail me here and I understood the internals of the library wrong.