kenjdavidson/react-native-bluetooth-classic #337: Manage bluetooth permissions on IOS

avanmalleghem, 2024-08-07:
**Mobile Device Environment**
iOS

**Application Environment**
Provide information about your development environment:
- React Native version: 0.74
- RN Bluetooth Classic version: 1.73.0

**Describe the bug**
Connection flow works perfectly on my Android phone. To do so, I added mainly 2 things : 
- In app.json : 

```
    "android": {
      "permissions": [
        "android.permission.BLUETOOTH_CONNECT"
      ],
     [....]
    },
```
- In App.tsx (my "main"), code to ask for permissions : 

```
import { PermissionsAndroid } from 'react-native'
[...]
  const requestAccessFineLocationPermission = async () => {
    const granted = await PermissionsAndroid.request(
      PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION
    )
    return granted === PermissionsAndroid.RESULTS.GRANTED
  }

  const requestBluetoothPermission = async () => {
    const granted = await PermissionsAndroid.request(
      PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT
    )
    return granted === PermissionsAndroid.RESULTS.GRANTED
  }

  useEffect(() => {
    ;(async () => {
      let granted =
        (await requestAccessFineLocationPermission()) && (await requestBluetoothPermission())

      if (!granted) {
        console.error('Access fine location or Bluetooth was not granted')
      }
    })()
}, [])
```

Unfortunately, on IOS, it crashes directly without anything in the terminal. How can I make it work, I guess it is a permission issue but which one ? Here is the stacktrace I receive : 
```
Thread 0 Crashed:
0   app                          	0x0000000100e6db1c Swift runtime failure: Unexpectedly found nil while unwrapping an Optional value + 0 (<compiler-generated>:0)
1   app                          	0x0000000100e6db1c RNBluetoothClassic.init() + 948 (RNBluetoothClassic.swift:69)
```

kenjdavidson, 2024-08-07:
This is most likely to do with the MFi protocols that we were talking about before.   What bluetooth device are you using? Do you have the MFi protocols for it?   Without these the IOS part of this library is useless to you.

avanmalleghem, 2024-08-07:
I'm on an Iphone8. I don't know how to answer to your question (I don't know if it is what you expect as answer). I don't think it is related to the device I want to connect to because the flow crashes before trying to reach the device.

kenjdavidson, 2024-08-07:
Have you read this section of the documentation

https://kenjdavidson.com/react-native-bluetooth-classic/ios/#mfi-protocols

it documents what is expected in terms of updated the `plist` file with the appropriate protocols.   These protocols are required by IOS to communicate with the device.   They are essentially secrets that tell IOS that your device has been configured and allowed through Bluetooth classic.

kenjdavidson, 2024-08-07:
For example, here is a super awesome document on how annoying it is to work with the Zebra protocol strings

https://www.zebra.com/content/dam/zebra_dam/en/faq/ios-app-faq-whitelisting-en-us.pdf

kenjdavidson, 2024-08-07:
Unless you absolutely need to use Bluetooth Classic and MFi devices, I would suggest not using this library, and using a BLE library instead.

avanmalleghem, 2024-08-08:
Oh wow... Took time to process the information you provided to me but it's clear now. For those reading it, SPP/RFCOMM (bluetooth classic protocol) is not supported on iOS without doing a certification for your custom device you want to connect your iOS to to use MFi protocols. If it is not a custom device you are created, maybe the device you want to use is MFi compliant.

=> I'm gonna try BLE first and see if I can reach expected performances.

closed 2024-08-08 by avanmalleghem