kenjdavidson/react-native-bluetooth-classic #339: functions are not working

vzan2012, 2024-08-22:
**Application Environment**
Provide information about your development environment:
- React Native version: 0.74.5
- RN Bluetooth Classic version: 1.0.5

Hello, I'm using react native with Expo and I have followed your docs for the expo plugin configs,

I'm able to get data for the device variables. 

But I'm unable to use the functions, getting errors of undefined.

[image]

kenjdavidson, 2024-08-26:
Sorry I missed this, is this android/ios?  `isConnected` should be available on both environments.  Is this the only function that isn't available, or is it the only one that you are using?

https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/7e5c889670c56fa87954aee3baff9001e21f5568/src/BluetoothDevice.ts#L70

which just calls through to 

https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/7e5c889670c56fa87954aee3baff9001e21f5568/src/BluetoothModule.ts#L117

Can you provide some more context of what type of object `device` is in. that instance?
What functions are available on it?

You'll need to debug a little.

vzan2012, 2024-08-26:
Hello @kenjdavidson ,

I'm using react-native (using expo). 

This is my code for your reference

`import { Alert } from "react-native";
import RNBluetoothClassic, {
  BluetoothDevice,
} from "react-native-bluetooth-classic";
import { TBluetoothOtherDevice } from "./BluetoothTypes";

export const scanOtherDevices = async (): Promise<BluetoothDevice[]> => {
  try {
    if (!RNBluetoothClassic) {
      throw new Error("RNBluetoothClassic is not initialized");
    }

    // const available = await RNBluetoothClassic.requestBluetoothEnabled();

    const devices = await RNBluetoothClassic.startDiscovery();
    // console.log(devices);

    const filteredDevices = devices.filter((d) => {
      const device = d as TBluetoothOtherDevice;
      device["source"] = "react-native-bluetooth-classic";
      return !device.name.includes(":");
    });

    setTimeout(() => RNBluetoothClassic.cancelDiscovery(), 10000);

    return filteredDevices;
  } catch (error) {
    console.error("Error scanning for devices: ", error);
    return [];
  }
};`

Here, I get the list of scanned devices (including bluetooth headset devices, tvs, etc...)

But next step, when I try to connect the device (creating a function) ...

`export const connectToClassicDevice = async (
  device: BluetoothDevice | TCombinedDevice
) => {
  try {
    // const isConnectedDevice1 = await device.isConnected();
    const isDeviceBonded = await device.bonded;
    console.log("Device: ");

    // console.log(isConnectedDevice1);

    // if (!isDeviceBonded) {
    //   Alert.alert(AppConfig.name, "Device is not bonded");
    // }

    // const connectedDevice = await device.connect();
    console.log("DN :" + device.name);
    console.log("DN ADR: " + device.address);
    console.log("DN Bonded: " + isDeviceBonded);
    // console.log("CD: " + connectedDevice);
    // console.log("isConnectedDevice: " + isConnectedDevice);
    console.log("****");
  } catch (error) {
    console.log(`Failed to connect the device: ${device.name}`);
    console.error(error);
  }
};
`

Here in the above code...

I'm able to print device name (variables), but I try to access the functions I get the error like either **undefined** or **its not a function**

vzan2012, 2024-08-26:
Actually I was trying to connect the device but it was failing.

So I started to debug line by line 
and I tried to print the variables creating with try catch to find the exact error.

I get bug exactly at the functions....

 `// const isConnectedDevice1 = await device.isConnected();`
 
[image]
 

For the other types... 
`TCombinedDevice?
TBluetoothOtherDevice?`

I getting data from other ble library, but currently I don't pass the data (Its commented on the another part of the file)

vzan2012, 2024-08-26:
For example, when I try to access the functions...

[image]

I get the similar error...
**[TypeError: device.XXXXX() is not a function (it is undefined)]**

kenjdavidson, 2024-08-26:
While I appreciate the screenshots of the console, you really need to breakpoint the line and show what kind of object that actually is?

Is it a `BluetoothDevice` object and actually doesn't have those functions.  If this were the case, there would be a bunch of issues related to that coming in (I would assume).

You're calling

```
export const connectToClassicDevice = async (
  device: BluetoothDevice | TCombinedDevice
) => {
```

But you're not showing where you're calling that, or what the actual object is that you're passing in.   Javascript is so shitty that it could be fine during compile time but completely different during runtime, so you'll need to debug it.

kenjdavidson, 2024-08-26:
Well, based on that it sure looks like the `BluetoothDevice` it's got both `_bluetoothModule` and `_nativeDevice`.   Not really sure why it doesn't have the functions associated to that class. 

You'll need to do a little more debugging to find out where that object is created.

vzan2012, 2024-08-26:
Hello @kenjdavidson,

I could not find the error.

But do you have an working example (link) implemented.

So that it will be easy for me to understand and debug it.

kenjdavidson, 2024-08-26:
Not sure what you mean, there is a working example in the `react-native-bluetooth-classic-apps` project (along side this one) which is used (or at least what I used when I actively did development) to verify and run the library.  You can take a look at it and see what's up.

There are also a large number of projects that use this library, that you can access and look at from the main page (right side panel).

vzan2012, 2024-08-26:
Yes, Thank you for providing **react-native-bluetooth-classic-apps** repo.

https://github.com/kenjdavidson/react-native-bluetooth-classic-apps

I don't know about this repo.

Let me check the my code with the example and I can able to bug in my code.

vzan2012, 2024-08-28:
Hello @kenjdavidson,

After analyzing the examples, I changed my implementation so I used 

`

export const connectToClassicDeviceWithRetries = async (
  device: BluetoothDevice | TCombinedDevice,
  retries = 3
) => {
  let attempt = 1;

  try {
    console.log(
      `Atempt: ${attempt} to connect device - ${device.name}, address: ${device.address}`
    );
    const connection = await RNBluetoothClassic.connectToDevice(
      device.address,
      {
        connectorType: "rfcomm",
        delimiter: "\n",
        charset: Platform.OS === "ios" ? 1536 : "utf-8",
      }
    );

    if (connection) {
      console.log("Connection successful");
      console.log(connection);
      return;
    }
  } catch (error) {
    console.error(`Failed to connect on attempt: ${attempt}`, error);
    attempt++;

    if (attempt >= retries) {
      console.error(`Max retries reached. Could not connect to the device`);
    }
  }
};

export const connectToClassicDevice = async (
  device: BluetoothDevice | TCombinedDevice
) => {
  try {
    const isDeviceBonded = await device.bonded;
    console.log("Device: ");

    // const connectedDevice = await device.;
    console.log("DN :" + device.name);
    console.log("DN ADR: " + device.address);
    console.log("DN Bonded: " + isDeviceBonded);

    const isConnected = await RNBluetoothClassic.isDeviceConnected(
      device.address
    );

    if (isConnected) {
      console.log("Device is already connected");
      return;
    } else {
      console.log("Device is not connected, can be connected");
    }

    await connectToClassicDeviceWithRetries(device);
  } catch (error) {
    console.log(`Failed to connect the device: ${device.name}`);
    console.error(error);
  }
};
`

But at the end, I'm unable to connect the device. 

I get with an error,

[image]

Here I don't know to resolve it.

Is there any possible solutions for the error ?

Please let me know.

Thank you.

kenjdavidson, 2024-08-28:
The error says that the socket isn't open, most likely your device isn't accepting or ready to connect.

image.png (view on web)<https://github.com/user-attachments/assets/e0dfae82-89c1-4156-956a-7f8310b76070>

vzan2012, 2024-08-28:
Is there any possible alternate solution to debug ?

Because I tried to connect with another device but getting the same error.

[image]

kenjdavidson, 2024-08-28:
Besides debugging... No.

You need to ensure the other devi e is set to accept. Not sure what device you're trying to connect with.

image.png (view on web)<https://github.com/user-attachments/assets/d7d2959d-ac0c-4de2-88f7-0b6cf6a705b5>

vzan2012, 2024-08-28:
I was trying to connect with Bluetooth headset and Bluetooth speakers.

The main goal of the app, is to connect the bluetooth devices (headset and speakers) and control them like (adjusting the volumes and getting battery status)

kenjdavidson, 2024-08-28:
Bluetooth headset and speakers usually use the H2soemthing profile, not the SPP profile.  You'll need to actually put the devices into accept mode and attempt to connect them, but it probably won't happen.   You need to read a lot more about the bluetooth profiles available, this library may not be the best choice for what you're doing.

closed 2024-08-28 by kenjdavidson

vzan2012, 2024-08-28:
Hello @kenjdavidson,

Thank you for letting me know. 

I didn't know about the specific bluetooth profile and I decided to go on with another better library like "react-native-ble-plx" or "react-native-ble-manager".

kenjdavidson, 2024-08-28:
Those are two completely different libraries that survey different purposes. But I appreciate your thoughts on them being better.

Glad you got it working.