kenjdavidson/react-native-bluetooth-classic #341: [Android] [Crashed] App crashed after update the device setting

thnhat51202574, 2024-10-30:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Android - Samsung S22 - Android 14
- OS: Android 14

**Application Environment**
Provide information about your development environment:
- React Native version: 0.72.19
- RN Bluetooth Classic version: "^1.73.0-rc.9"

**Describe the bug**
- AppCrashed after change the deviec setting

**To Reproduce**
- Prerequisite: Bluetooth on the device is turned on.
- Steps:
1. Open the Android app at the verify-first screen (the user might not allow Bluetooth permission).
2. Go to device settings to change the language.

- The reason:
function `onHostDestroy` is trigered and throw the exception.
```
    @Override
    public void onHostDestroy() {
        if (BuildConfig.DEBUG)
            Log.d(TAG, "onHostDestroy: stop discovery, connections and unregister receivers");
        mAdapter.cancelDiscovery();
    }
```

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.

kenjdavidson, 2024-10-30:
Error looks like you need to add the `BLUETOOTH_SCAN` permission to the application.  Is there something else that is showing up in the logs that leads you to believe this is not a configuration issue?

thnhat51202574, 2024-10-30:
Hi @kenjdavidson, thank you for responding.
These are the permissions we added to the `AndroidManifest.xml`:
```
<uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" tools:targetApi="s" />
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
````
The issue never happens if the user accepts/allows the permission request.
However, if the user opens the app for the first time (without taking any additional action) and then changes the device settings, we encounter a crash

kenjdavidson, 2024-10-30:
That sounds about right, there is information on the React Native site (and maybe these docs) saying that you need to manage that, you shouldn't be attempting to access anything without requesting permission.  You should be able to handle that by wrapping all the bluetooth logic in a wrapper that checks if they've already give permission.

kenjdavidson, 2024-10-30:
https://kenjdavidson.com/react-native-bluetooth-classic/android/#android-permissions

If you feel like this needs more information, once you sort it out feel free to open a PR updating the docs for future users.

thnhat51202574, 2024-10-31:
@kenjdavidson 
We have handled the permission request in our app, but it doesn't seem to be relevant. 
The issue occurs immediately after installing `react-native-bluetooth-classic` without adding any additional code. 

The `onHostDestroy` function is registered in the Android codebase after running the `yarn install` command

kenjdavidson, 2024-10-31:
If you figure it out, please open a PR and we can get the fix in.   I'm not sure there is much I can do as I don't have access to this code nor do I really do Mobile/React Native development anymore - so I'm well out of the loop of what's going on.  I'm really just maintaining the library as others use it.

If I get some time, I can look around for docs and issues, but at this point you're going to need to do a little debugging and try to work it out.

Sorry.

kenjdavidson, 2024-10-31:
I would start by wrapping the line in question https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/7e5c889670c56fa87954aee3baff9001e21f5568/android/src/main/java/kjd/reactnative/bluetooth/RNBluetoothClassicModule.java#L361 with some kind of check:
- to see that the user has provided Bluetooth permissions. 
- or only do the cancel if discovery has been started, since you'd need permissions to do so.