kenjdavidson/react-native-bluetooth-classic #343: no read() data is received

Code-Divine, 2024-12-15:
"react-native": "0.76.5",
"react-native-bluetooth-classic": "^1.73.0-rc.12",

Device Arduino bluetooth HC-06

The command is successfully sent and executed but afterwards .available() returns zero and .read() returns nothing. It blocks my device because the device itself keeps the output in the buffer and doesn't accept new commands until is cleared. So I have to use another 3rd party program that reads the output buffer and then I can send commands with this plugin again.

Here is a sample of my code

Reading method:

const performRead = async (device) => {
    try {
      console.log("Polling for available messages");
      let available = await device.available();
      console.log(`There is data available [${available}], attempting read`);

      if (available > 0) {
        for (let i = 0; i < available; i++) {
          console.log(`reading ${i}th time`);
          let data = await device.read();
          console.log(`Read data ${data}`);
          console.log(data);
        }
      } else {
        console.log("No read data available");
      }
    } catch (err) {
      console.log(err);
    }
  };
  
  Main method:

const connectAndSendData = async () => {
    try {
      console.log("connecting");
      let device = await RNBluetoothClassic.connectToDevice(
        selectedDevice.address
      );
      if (await device.isConnected()) {
        console.log("device", device);
        const readInterval = setInterval(() => performRead(device), 5000);
        const write = await device.write(
          createCommand(1)
        );
        await new Promise((resolve) => setTimeout(resolve, 20000));
        console.log("disconecting", disconnect);
        clearInterval(readInterval);
        const disconnect = await device.disconnect();
        console.log("disconnected", disconnect);
      }
    } catch (err) {
      if (await device.isConnected()) {
        const disconnect = await device.disconnect();
        console.log("disconnected in catch", disconnect);
      }
      console.log("error", err.message);
      setError(err.message);
    }
  };

kenjdavidson, 2024-12-16:
The majority of the time the issue with Arduino is the `delimiter` used for reading and writing.  Please ensure that you're matching the `delimiter` between the two.   You're connecting to the device without any parameters, which means the default `delimiter` is going to be `\n` make sure the writing to Arduino is not `\r\n`.

https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/7e5c889670c56fa87954aee3baff9001e21f5568/android/src/main/java/kjd/reactnative/bluetooth/conn/StandardOption.java#L43

Other than that, you'll need to perform the debugging yourself.  My development on this project is limited and I've been relying on the community for assistance in maintaining since I no longer work with this library on a daily basis.

Code-Divine, 2024-12-16:
That helped in fact. The problem was in fact the delimiter.

Anyone having problem with no data in .read() with Arduino devices is because Arduino uses no line endings. You basically have to do 

```
let device = await RNBluetoothClassic.connectToDevice(
        selectedDevice.address,
        {
          delimiter: "",
        }
      );
```

and voila, it works.

closed 2024-12-16 by Code-Divine