kenjdavidson/react-native-bluetooth-classic #345: difficulty with read()

Giacomo90, 2025-02-02:
Hi guys,
 I have a problem I'm creating an app where I can communicate via bluetooth using base64 strings.
I have 2 devices where on the first one I use device.write("string") so far all good, the method works and sends the string,
device.read() instead inside a while loop always returns null and never the value sent by write(). How do I solve it?

kenjdavidson, 2025-02-04:
You'll need to provide a little of your own debugging for assistance, there are a number of resolved issues and discussions documenting how to work with the Base64 strings for communication.

Giacomo90, 2025-02-04:
Of course in the meantime where do I find these discussions?

Giacomo90, 2025-02-05:
they didn't help me much, I'll upload my 2 codes for you.

```  async function sendDataAudio(bluetooth){
try {
  const ISconn:boolean = await bluetooth.isConnected();

  if(ISconn){
    ScreenAudioRecorder.on('data',async (data:string)=>{
      await bluetooth.write(data);

  });

 } else { ToastAndroid.show("Disconnesso",2500);
   setIsConnection("In attesa di una connessione bluetooth")
  bluetooth.disconnect()}
 } catch (error) {
   console.error(error);
   }
                }

```
as said in the first comment, this works fine, strings are sent correctly, I did a test with bluetooth Classic apk.

```
    async function connectDevice() {
        try {
        const _dvc= await RNBluetoothClassic.connectToDevice(props.address)
        if(_dvc){
    
const msg = await _dvc.read()
if(msg){console.log(msg)}
        } 
       
  
        } catch (error) {
          console.error("Error connecting to device:");
        }
      }
```
_dvc connects without problems, but read() always returns null and does not read the strings from device 1

kenjdavidson, 2025-02-05:
So the first code is device 1, and this is writing to the device from the ScreenAudioRecorder?  I'm assuming that device 1 is using the `accept` at some point in time?

The second code is device 2, which is attempting to connect and read?

My first question is always the same, are you working with the right delimiter on both the sender and the receiver?   I cannot see your accept code, but your connect code uses the default `\r`.  If you're not sending that, there is no delimitation and therefore you will never get messages.

Also, I would not continually connect to and disconnect from the device, as this doesn't make much sense.  You probably want to connect once and then either poll or listen for the On Read event.

Giacomo90, 2025-02-05:
Yes the first code is device 1 and the second is device 2.
 No I am not using delimiters on either device.
```
   const optBlt ={
  uuid:"00001101-0000-1000-8000-00805F9B34FB"
   }

  setIsConnection("In attesa di una connessione bluetooth");
  RNBluetoothClassic.accept(optBlt)
  .then((devices)=>{sendDataAudio(devices); setIsConnection(`Connesso con ${devices.name}`)})
  .catch((err)=>{   
     setIsConnection("Impossibile stabilire una connessione bluetooth");
    ToastAndroid.show("Si é verificato un errore",3000);
    console.log(err)})

```

the connection on device 2 instead I wrote in the previous comment

Giacomo90, 2025-02-05:
solved by adding "\r\n" to the end of each base64 string.

Example:
```
var base64mod = base64 +"\r\n"

devices.write(base64mod)

```

thank you so much for your help

closed 2025-02-05 by kenjdavidson