kenjdavidson/react-native-bluetooth-classic #346: Write command not working on react native 0.76.6

mMarcos208, 2025-02-05:
**Mobile Device Environment**

- Device: Android 8/13

**Application Environment**

- React Native version: "0.76.6"
- RN Bluetooth Classic version: ^1.73.0-rc.13

**Describe the bug**
When I send a write command to device, device does not respond to me. In the app **Serial Bluetoth Terminal** works well.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to device
2. Send write command to device
3. onDataReceived not working

**Expected behavior**
Device send me the response.

Code: 

```
  const connect = async () => {
    const deviceInfo = await RNBluetoothClassic.connectToDevice('24:4C:AB:37:14:EE')

    deviceInfo.onDataReceived((event) => {
      console.log('event', event.data)
    })

    console.log('device connected...')
  }
```

```
const readFromDevice = async () => {
    const buffer = Buffer.from(';peso')
    const deviceInfo = await RNBluetoothClassic.getConnectedDevice('24:4C:AB:37:14:EE')
    await deviceInfo.write(buffer)

    setTimeout(() => {
      deviceInfo
        .read()
        .then((data) => {
          console.log('Data recevied:', data)
        })
        .catch((error) => {
          console.error('Error', error)
        })
    }, 1024)
}
```

kenjdavidson, 2025-02-05:
You're using both `onDataReceived` and `read` functions? You shouldn't be using both at the same time as it could cause problems, you should only be using one.

With that said, you'll need to do some debugging on your side to determine what's going on in the native code.  As I'm sure you've read, I don't have much time to spend on issues, and every time I end up firing up my local application it works.  So you'll need to provide some of your own debugging, and if you find an issue I'll gladly assist and accept a PR.

kenjdavidson, 2025-02-05:
Just initially, your write buffer doesn't have `\n\r` which is the default delimiter, without a delimiter the read methods won't work, I would look into the connection options and sending the appropriate message delimiter.

mMarcos208, 2025-02-05:
I'm not using _onDataReceived_ and read _together_. I'm testing with \n and \r and not working :(

I have other devices that work well.

kenjdavidson, 2025-02-05:
Based on your connect and read from device code you're using them together. You're write code also has no delimiter.  Would be helpful if you posted the actual code instead of functions you're not using, so I can give appropriate advice or suggestions.

Please provide some debugging information and maybe we can sort it out.

mMarcos208, 2025-02-05:
I'm testing with   

`const buffer = Buffer.from(';peso\r')` and works. Thank you so much.

kenjdavidson, 2025-02-05:
Welcome.

closed 2025-02-05 by kenjdavidson