kenjdavidson/react-native-bluetooth-classic #351: Suggest create the event which named "onFoundDevice" for after executing "startDiscovery"

sunhanA, 2025-03-31:
**Mobile Device Environment**
On which device(s) would you like this feature to be implemented.   If Android only, the IOS implementations must be created as well - keep not of this for your pull requests.
- Device: [e.g. Android, iPhone6]
- OS: [e.g. API19, Nouget, iOS8.1]

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.  If you've researched the available libraries, feel free to add notes or implementation details as you see fit.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

kenjdavidson, 2025-03-31:
There were some issues with the device discovery events that were causing problems, so I removed them in favour of just waiting for the discovery promise to resolve.  If you'd really like those back, I would suggest working on them and opening a PR.  As long as they don't cause issues, they can be re-added, sadly I don't remember what those issues are.

There is `onDeviceDiscovered` which I think you mean instead of `onFoundDevice` (maybe, I can't find that looking quickly at the docs).

It also does look like it's being called `DiscoveryReceiver` 

https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/cabeefba3d17192471dcdb708bf9ddc063c81d17/android/src/main/java/kjd/reactnative/bluetooth/receiver/DiscoveryReceiver.java#L55

Are you saying this isn't working?  Can you provide some debugging?

kenjdavidson, 2025-03-31:
For whatever reason, looks like it's being called twice

```
                mCallback.onDeviceDiscovered(found);
                unpairedDevices.put(device.getAddress(), found);

                mCallback.onDeviceDiscovered(found);
```

which probably doesn't need to happen.

kenjdavidson, 2025-03-31:
It's also possible for a device to be found multiple times, with the differing RSSI values, that could also be a feature that is added.

sunhanA, 2025-04-01:
thanks, I have a try