kenjdavidson/react-native-bluetooth-classic #352: iOS 18 crashes the app on device.write()

PietroGranati, 2025-05-08:
**Mobile Device Environment**
iPad with iOS 18.4.1

**Application Environment**
- React Native version:0.73.6
- RN Bluetooth Classic version: last

**Describe the bug**
Since the update to iOS18 some thermal printers (Seiko by now) started to made app crash on device .write()
the only fix is to use iOS < 18
Some other printers work (Brother) I'm currently investigating the issue

kenjdavidson, 2025-05-08:
Appreciate you looking into it, if you have any thoughts feel free to run them by me.  I have neither an IOS 18 device nor an MFI Compliant device, so I'm pretty much useless with regards to helping on the IOS side of things these days.

PietroGranati, 2025-07-24:
Fixed!
I've modified **ios/conn/DelimitedStringDeviceConnectionImpl** file, under the read method:

`func read() -> String? {
        NSLog("(BluetoothDevice:readFromDevice) Reading device %@ until delimiter %@",
              self.accessory.serialNumber, self.delimiter)

        //let content = String(data: inBuffer, encoding: self.encoding)!
        guard let content = String(data: inBuffer, encoding: self.encoding) else {
            NSLog("Error, impossible buffer decode")
            return nil
        }
        var message:String?

        if (delimiter.count == 0) {
            message = content
            inBuffer.removeAll()
        } else {
            if let index = content.index(of: self.delimiter) {
                message = String(content[..<index])
                inBuffer = String(content[content.index(after: index)...])
                    .data(using: self.encoding) ?? Data()
            }
        }

        return message
    }`

kenjdavidson, 2025-07-24:
Sweet. Want to open a pr? And I'll get it in and released?

closed 2025-11-04 by kenjdavidson