kenjdavidson/react-native-bluetooth-classic #355: Crash BLUETOOTH_SCAN permission

DanUnited, 2025-09-08:
### Reproduced only on Xiaomi or Redmi tablets
1. Add & autolinking react-native-bluetooth-classic
2. rebuild app & run on android
```
"react-native": "0.75.2",
newArchEnabled=false
ORG_GRADLE_PROJECT_newArchEnabled=false
```

Minimize app or close it -> CatLog:

```
Fatal Exception: java.lang.RuntimeException
Unable to destroy activity {MyAppName.MainActivity}: java.lang.SecurityException: Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@87c28247: AdapterService cancelDiscovery

 Fatal Exception: java.lang.RuntimeException: Unable to destroy activity {MyAppName.MainActivity}: java.lang.SecurityException: Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@87c28247: AdapterService cancelDiscovery
       at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:6430)
       at android.app.ActivityThread.handleDestroyActivity(ActivityThread.java:6469)
       at android.app.servertransaction.DestroyActivityItem.execute(DestroyActivityItem.java:46)
       at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:60)
       at android.app.servertransaction.TransactionExecutor.executeLifecycleItem(TransactionExecutor.java:237)
       at android.app.servertransaction.TransactionExecutor.executeTransactionItems(TransactionExecutor.java:110)
       at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:84)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2875)
       at android.os.Handler.dispatchMessage(Handler.java:107)
       at android.os.Looper.loopOnce(Looper.java:249)
       at android.os.Looper.loop(Looper.java:337)
       at android.app.ActivityThread.main(ActivityThread.java:9608)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:615)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
```

DanUnited, 2025-09-08:
# Security Analysis: react-native-bluetooth-classic Unsafe Bluetooth Calls

## Issue Summary
The `react-native-bluetooth-classic` package contains unsafe Bluetooth API calls that cause security exceptions when the Android activity is destroyed, specifically related to missing `BLUETOOTH_SCAN` permission checks.

## Error Context
```
Unable to destroy activity {com.indicaonline.pos/com.indicaonline.pos.MainActivity}:
java.lang.SecurityException: Need android.permission.BLUETOOTH_SCAN permission for
android.content.AttributionSource@87c28247: AdapterService cancelDiscovery
```

## Critical Vulnerabilities Found

### 1. Unsafe `cancelDiscovery()` in Activity Lifecycle (HIGH RISK)

**Location:** `RNBluetoothClassicModule.java:361`
```java
@Override
public void onHostDestroy() {
    if (BuildConfig.DEBUG)
        Log.d(TAG, "onHostDestroy: stop discovery, connections and unregister receivers");
    
    mAdapter.cancelDiscovery(); // ⚠️ UNSAFE: No permission check
}
```

**Risk:** Direct cause of the security exception. Called automatically when activity is destroyed.

### 2. Unsafe `cancelDiscovery()` in Receiver Cleanup (HIGH RISK)

**Location:** `RNBluetoothClassicModule.java:1117`
```java
private void unregisterBluetoothReceivers() {
    // ... other cleanup code ...
    
    if (mDiscoveryReceiver != null) {
        mAdapter.cancelDiscovery(); // ⚠️ UNSAFE: No permission check
        getReactApplicationContext().unregisterReceiver(mDiscoveryReceiver);
        mDiscoveryReceiver = null;
    }
}
```

**Risk:** Called during `onHostPause()` and `onHostDestroy()` lifecycle events.

This match solved this issue:

### Patch for Android 12+ Bluetooth permission handling

This patch introduces a `safeCancelDiscovery()` method with proper permission checks for Android 12+ (`BLUETOOTH_SCAN`), replacing direct calls to `mAdapter.cancelDiscovery()`.

```diff
diff --git a/node_modules/react-native-bluetooth-classic/android/src/main/java/kjd/reactnative/bluetooth/RNBluetoothClassicModule.java b/node_modules/react-native-bluetooth-classic/android/src/main/java/kjd/reactnative/bluetooth/RNBluetoothClassicModule.java
index 56a30b6..95a1f82 100644
--- a/node_modules/react-native-bluetooth-classic/android/src/main/java/kjd/reactnative/bluetooth/RNBluetoothClassicModule.java
+++ b/node_modules/react-native-bluetooth-classic/android/src/main/java/kjd/reactnative/bluetooth/RNBluetoothClassicModule.java
@@ -9,10 +9,13 @@ import android.content.ActivityNotFoundException;
 import android.content.BroadcastReceiver;
 import android.content.Intent;
 import android.content.IntentFilter;
+import android.content.pm.PackageManager;
 import android.os.Build;
 import android.provider.Settings;
 import android.util.Base64;
 import android.util.Log;
+import android.Manifest;
+import androidx.core.app.ActivityCompat;
 
 import androidx.annotation.NonNull;
 
@@ -280,6 +283,27 @@ public class RNBluetoothClassicModule
     private boolean checkBluetoothAdapter() {
         return (mAdapter != null && mAdapter.isEnabled());
     }
+
+    /**
+     * Safely cancels Bluetooth discovery with proper permission checks for Android 12+
+     */
+    private void safeCancelDiscovery() {
+        try {
+            if (mAdapter != null && mAdapter.isEnabled()) {
+                // Check if we have required permission for API 31+
+                if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.S) {
+                    if (ActivityCompat.checkSelfPermission(getReactApplicationContext(), 
+                            Manifest.permission.BLUETOOTH_SCAN) != PackageManager.PERMISSION_GRANTED) {
+                        Log.w(TAG, "BLUETOOTH_SCAN permission not granted, skipping cancelDiscovery");
+                        return;
+                    }
+                }
+                mAdapter.cancelDiscovery();
+            }
+        } catch (SecurityException e) {
+            Log.e(TAG, "SecurityException in cancelDiscovery: " + e.getMessage());
+        }
+    }
     // endregion
 
     // region: ActivityEventListener
@@ -358,7 +382,7 @@ public class RNBluetoothClassicModule
         if (BuildConfig.DEBUG)
             Log.d(TAG, "onHostDestroy: stop discovery, connections and unregister receivers");
 
-        mAdapter.cancelDiscovery();
+        safeCancelDiscovery();
     }
     // endregion
 
@@ -1114,7 +1138,7 @@ public class RNBluetoothClassicModule
         }
 
         if (mDiscoveryReceiver != null) {
-            mAdapter.cancelDiscovery();
+            safeCancelDiscovery();
             getReactApplicationContext().unregisterReceiver(mDiscoveryReceiver);
             mDiscoveryReceiver = null;
         }

kenjdavidson, 2025-09-08:
Feel free to open a PR with this.

closed 2025-11-04 by kenjdavidson